aliyun/iotkit-embedded #2: error: MQTT read error

WeldonWangwang, 2017-10-18:
Hi,

I used IoT-SDK on other platform, and the device connected to the net by WiFi,
and I used the example `mqtt-example`, but I got the following error after connecting to the iot controler.

```
[inf] utils_network_ssl_read(236): ssl recv error: code = -30848, err_str = 'SSL - The peer notified us that '
[dbg] iotx_mc_read_packet(746): mqtt read error, rc=-2
[dbg] iotx_mc_cycle(1214): readPacket error,result = -1
[err] iotx_mc_keepalive(1798): network is disconnected!
event_handle|060 :: MQTT disconnect.
[inf] utils_network_ssl_disconnect(297): ssl_disconnect
[err] iotx_mc_publish(1420): mqtt client state is error,state = 4
mqtt_client|232 :: error occur when publish
[err] iotx_mc_unsubscribe(1386): mqtt client state is error,state = 4
[inf] iotx_mc_release(2100): mqtt release!
```
I want to know if it's the problem of WiFi or other reasons, thank you!

tongjiang, 2017-10-19:
Do u have ported the ssl /tcp HAL API on your platform?
The iotkit uses mbedtls only as a example for tls/dtls, you should realize it yourself.

edwardyangx, 2017-10-23:
`
src/import/linux/include/mbedtls/ssl.h
#define MBEDTLS_ERR_SSL_PEER_CLOSE_NOTIFY                 -0x7880
`

I saw `code = -30848` which means -0x7880, so looks to me in this case SSL connection was closed and SDK you ported had to disconnect MQTT session

WeldonWangwang, 2017-10-24:
@tongjiang @edwardyangx , thanks,

I added a small delay after `mbedtls_ssl_readutils` in the `_network_ssl_read`, so I guess it start return without get the packages.

edwardyangx, 2017-10-26:
@WeldonWangwang That's a valuable feedback, which proved again HAL_TLS_mbed.c on different platforms all need corresponding adjustment or even re-implentation.

Well done. I will close this issue then.

closed 2017-10-26 by edwardyangx