aliyun/iotkit-embedded #33: ota模块下载过程中为什么一直要httpclient_send_request?

chipenglau, 2018-01-26:
在ota_mqtt-example.c中
```
do {

    len = IOT_OTA_FetchYield(h_ota, buf_ota, OTA_BUF_LEN, 1);
    if (len > 0) {
        if (1 != fwrite(buf_ota, len, 1, fp)) {
            EXAMPLE_TRACE("write data to file failed");
            rc = -1;
            break;
        }
    }

    /* get OTA information */
    IOT_OTA_Ioctl(h_ota, IOT_OTAG_FETCHED_SIZE, &size_downloaded, 4);
    IOT_OTA_Ioctl(h_ota, IOT_OTAG_FILE_SIZE, &size_file, 4);
    IOT_OTA_Ioctl(h_ota, IOT_OTAG_MD5SUM, md5sum, 33);
    IOT_OTA_Ioctl(h_ota, IOT_OTAG_VERSION, version, 128);

    last_percent = percent;
    percent = (size_downloaded * 100) / size_file;
    if (percent - last_percent > 0) {
        IOT_OTA_ReportProgress(h_ota, percent, NULL);
        IOT_OTA_ReportProgress(h_ota, percent, "hello");
    }
    IOT_MQTT_Yield(pclient, 100);
} while (!IOT_OTA_IsFetchFinish(h_ota));
```
为什么`IOT_OTA_FetchYield`要内部要一直调用`ofc_Fetch`->`httpclient_common`->`httpclient_send_request`，这样一个http/https请求要一直写数据，是不是很影响设备的性能，特备是性能受限的设备？

seblee, 2018-01-27:
這是OTA，就是從服務器上把固件down下來，直到數據全部down完，所以一直while，更新固件的時候應該不考慮設備使用吧，把數據完整down下來才是重要的，還有正常使用的時候固件不會經常更新才對

lifang870603, 2018-02-12:
一直调用httpclient_send_request  确实是一个bug，这个部分已经在最新的版本上修复。

非常感谢您提出的反馈，谢谢！

closed 2018-03-09 by lifang870603