aliyun/iotkit-embedded #71: make sdk时出现/usr/bin/ld: cannot find -liot_sdk

aauuaa, 2018-06-07:
aua@AUA:~/aliIot/iotkit-embedded$ make
SELECTED CONFIGURATION:

VENDOR :   ubuntu
MODEL  :   x86

CONFIGURE .............................. [sample]
CONFIGURE .............................. [src/cmp]
CONFIGURE .............................. [src/coap]
CONFIGURE .............................. [src/cota]
CONFIGURE .............................. [src/dm]
CONFIGURE .............................. [src/fota]
CONFIGURE .............................. [src/http]
CONFIGURE .............................. [src/log]
CONFIGURE .............................. [src/mqtt]
CONFIGURE .............................. [src/ota]
CONFIGURE .............................. [src/platform]
CONFIGURE .............................. [src/sdk-tests]
CONFIGURE .............................. [src/shadow]
CONFIGURE .............................. [src/subdev]
CONFIGURE .............................. [src/system]
CONFIGURE .............................. [src/tfs]
CONFIGURE .............................. [src/tls]
CONFIGURE .............................. [src/utils]

BUILDING WITH EXISTING CONFIGURATION:

VENDOR :   ubuntu
MODEL  :   x86

Components: 

. sample                                            
. src/dm                                            
. src/cmp                                           
. src/http                                          
. src/platform                                          
. src/tls                                           
. src/log                                           
. src/system                                          
. src/cota                                          
. src/ota                                           
. src/sdk-tests                                          
. src/fota                                          
. src/tfs                                           
. src/shadow                                          
. src/subdev                                          
. src/mqtt                                          
. src/coap                                          
. src/utils                                          

[CC] aes.o                      <= aes.c                                    
                                   makefile
[CC] asn1parse.o                <= asn1parse.c                                    
                                   makefile
[CC] base64.o                   <= base64.c                                    
                                   makefile
[CC] bignum.o                   <= bignum.c                                    
                                   makefile
[CC] cipher.o                   <= cipher.c                                    
                                   makefile
[CC] cipher_wrap.o              <= cipher_wrap.c                                    
                                   makefile
[CC] ctr_drbg.o                 <= ctr_drbg.c                                    
                                   makefile
[CC] debug.o                    <= debug.c                                    
                                   makefile
[CC] entropy.o                  <= entropy.c                                    
                                   makefile
[CC] entropy_poll.o             <= entropy_poll.c                                    
                                   makefile
[CC] error.o                    <= error.c                                    
                                   makefile
[CC] md.o                       <= md.c                                    
                                   makefile
[CC] md_wrap.o                  <= md_wrap.c                                    
                                   makefile
[CC] net_sockets.o              <= net_sockets.c                                    
                                   makefile
[CC] oid.o                      <= oid.c                                    
                                   makefile
[CC] pem.o                      <= pem.c                                    
                                   makefile
[CC] pk.o                       <= pk.c                                    
                                   makefile
[CC] pkparse.o                  <= pkparse.c                                    
                                   makefile
[CC] pk_wrap.o                  <= pk_wrap.c                                    
                                   makefile
[CC] platform.o                 <= platform.c                                    
                                   makefile
[CC] rsa.o                      <= rsa.c                                    
                                   makefile
[CC] sha1.o                     <= sha1.c                                    
                                   makefile
[CC] sha256.o                   <= sha256.c                                    
                                   makefile
[CC] ssl_ciphersuites.o         <= ssl_ciphersuites.c                                    
                                   makefile
[CC] ssl_cli.o                  <= ssl_cli.c                                    
                                   makefile
[CC] ssl_cookie.o               <= ssl_cookie.c                                    
                                   makefile
[CC] ssl_tls.o                  <= ssl_tls.c                                    
                                   makefile
[CC] timing.o                   <= timing.c                                    
                                   makefile
[CC] x509.o                     <= x509.c                                    
                                   makefile
[CC] x509_crt.o                 <= x509_crt.c                                    
                                   makefile
[AR] libiot_tls.a               <= aes.o                                    
                                   asn1parse.o
                                   base64.o
                                   bignum.o
                                   cipher.o
                                   cipher_wrap.o
                                   ctr_drbg.o
                                   debug.o
                                   entropy.o
                                   entropy_poll.o
                                   error.o
                                   md.o
                                   md_wrap.o
                                   net_sockets.o
                                   oid.o
                                   pem.o
                                   pk.o
                                   pkparse.o
                                   pk_wrap.o
                                   platform.o
                                   rsa.o
                                   sha1.o
                                   sha256.o
                                   ssl_ciphersuites.o
                                   ssl_cli.o
                                   ssl_cookie.o
                                   ssl_tls.o
                                   timing.o
                                   x509.o
                                   x509_crt.o
[CC] HAL_OS_linux.o             <= HAL_OS_linux.c                                    
                                   makefile
[CC] HAL_TCP_linux.o            <= HAL_TCP_linux.c                                    
                                   makefile
[CC] HAL_UDP_linux.o            <= HAL_UDP_linux.c                                    
                                   makefile
[CC] HAL_DTLS_mbedtls.o         <= HAL_DTLS_mbedtls.c                                    
                                   makefile
[CC] HAL_TLS_mbedtls.o          <= HAL_TLS_mbedtls.c                                    
                                   makefile
[CC] HAL_TLS_itls.o             <= HAL_TLS_itls.c                                    
                                   makefile
[AR] libiot_platform.a          <= HAL_OS_linux.o                                    
                                   HAL_TCP_linux.o
                                   HAL_UDP_linux.o
                                   HAL_DTLS_mbedtls.o
                                   HAL_TLS_mbedtls.o
                                   HAL_TLS_itls.o
[CC] mqtt-example.o             <= mqtt-example.c                                    
                                   makefile
[LD] mqtt-example               <= mqtt-example.o                                    
/usr/bin/ld: cannot find -liot_sdk
/usr/bin/ld: cannot find -liot_sdk
collect2: ld 返回 1
make[3]: *** [mqtt-example] 错误 1
make[2]: *** [mqtt-example] 错误 2
make[1]: *** [sample] 错误 2
make: *** [sub-mods] 错误 2

yuanfeng5721, 2018-06-08:
在 iot.mk 中增加LDFLAGS += -lrt

wanghui0574xs, 2018-08-27:
@yuanfeng5721 加在第几行, 我加在第5行后, 
LDFLAGS += -liot_platform 的下一行, 还是会报错undefined reference to `clock_gettime'

kendy37, 2018-09-18:
解决了吗？我也遇到了，求指引。
@yuanfeng5721 @wanghui0574xs