aliyun/iotkit-embedded #77: [问题] 如何讲SDK移植到其他OS平台

zhaojuntao, 2018-07-05:
如何讲SDK移植到其他OS平台？如 FreeRTOS 等RTOS？

哪里可以找到相关的移植文档？

谢谢

zhaojuntao, 2018-07-06:
新建了一个 config 内容如下：

```
CONFIG_ENV_CFLAGS = -mcpu=arm968e-s -mthumb-interwork -mthumb -ffunction-sections -fdata-sections

OVERRIDE_CC      = arm-none-eabi-gcc
OVERRIDE_CXX     = arm-none-eabi-g++
OVERRIDE_AS      = arm-none-eabi-gcc
OVERRIDE_AR      = arm-none-eabi-ar
OVERRIDE_LINK    = arm-none-eabi-g++

# CONFIG_src/platform =
CONFIG_sample =
CONFIG_src/sdk-tests =
```

编译报错：

```
/ali/iotkit-embedded/src/tls/mbedtls-in-iotkit/library/net_sockets.c:32:2: error: #error "This module only works on Unix and Windows, see MBEDTLS_NET_C in config.h"
 #error "This module only works on Unix and Windows, see MBEDTLS_NET_C in config.h"
  ^~~~~
```

lingyun0129, 2018-09-10:
同问，如何将SDK移植到mtk feature phone 平台，config不知道怎么配

zhaojuntao, 2018-09-11:
我移植了一份通用平台的，参考我 GitHub ali-iotkit 仓库，希望可以帮到你

lingyun0129, 2018-09-11:
@zhaojuntao thank you ，我试试看，谢谢了

Neutree, 2018-09-11:
多看几个移植示例就会了哦

https://github.com/Ai-Thinker-Open/GPRS_C_SDK/tree/master/libs/aliyun

https://github.com/espressif/esp-aliyun

lingyun0129, 2018-09-13:
@zhaojuntao 通用平台是什么意思？与具体硬件平台无关吗？如果我要把SDK移植到MTK平台该如何操作？

zhaojuntao, 2018-09-13:
@lingyun0129 这个包只是应用层组件而已，不受硬件平台限制，我使用 scons 进行构建的，你只要把相关的文件添加的你的编译系统里就可以了

closed 2018-09-13 by zhaojuntao