aliyun/iotkit-embedded #88: IOT_MQTT_Yield 报错，同时导致所在的线程阻塞，非常严重的bug，而且没法规避

titawork, 2018-08-01:
[inf] iotx_mc_keepalive(1861): network is reconnected!
current time : Tue Jul 31 23:57:40 2018

++++++++++++MQTT reconnect.
[inf] _network_ssl_read(468): ssl recv error: code = -30848, err_str = ''
[dbg] iotx_mc_read_packet(744): mqtt read error, rc=-2
[dbg] iotx_mc_cycle(1239): readPacket error,result = -1
**[inf] IOT_MQTT_Yield(2415): error occur** 
[err] iotx_mc_keepalive(1871): network is disconnected!
current time : Tue Jul 31 23:59:23 2018

++++++++++++MQTT disconnect.
[inf] _network_ssl_disconnect(531): ssl_disconnect
[dbg] iotx_mc_cycle(1225): state = 4
**[inf] IOT_MQTT_Yield(2415): error occur** 
[inf] iotx_mc_handle_reconnect(2072): Waiting to reconnect...
[inf] iotx_mc_handle_reconnect(2078): start to reconnect

lifang870603, 2018-08-20:
ssl recv error: code = -30848, err_str =   表明当前的ssl出错了，需要检查一下ssl出错原因。
云端把SSL连接断开  The peer notified us that the connection is going to be closed
1. 设备端数据连接过于频繁，触发云端限流，断开设备。
请关闭设备，等待一段时间(5分钟以后)再发起连接
2. 有多个设备使用相同的product_key和device_name与云端建立连接，之前连接的设备被云端踢开。
3. 设备端保活出错，没有及时发送MQTT ping packet （出错通常在连接建立1分钟以后）。
如果是保活问题，需要进一步分析。
4. 如果从未连接成功过，可以考虑是不是大小端口不匹配。
目前SDK 默认是小端，如果在大端环境上工作，请添加定义全局编译条件   REVERSED。

qiubin2016, 2021-05-13:
遇到相同问题，yield();这个函数会卡住几十秒。现在publish和yield放在同一个线程，yield阻塞几十秒，导致消息无法及时publish到IOT平台。

[DBG] iotx_mqtt_client.c:1535 [2021-04-27 15:38:12.475] PUBACK
[DBG] Iot.c:56 [2021-04-27 15:38:12.476] publish success, packet-id=184

[DBG] HAL_TLS_mbedtls.c:650 [2021-04-27 15:40:01.067] ssl recv error: code = -30848, err_str = 'SSL - The peer notified us that '
[DBG] iotx_mqtt_client.c:541 [2021-04-27 15:40:01.068] mqtt read error, rc=-2
[DBG] iotx_mqtt_client.c:1511 [2021-04-27 15:40:01.068] readPacket error,result = -14
[DBG] CIot.cpp:448 [2021-04-27 15:40:01.069] =============================================Yield time=94263ms
[DBG] iotx_mqtt_client.c:3099 [2021-04-27 15:40:01.073] mqtt client state is error,state = 3
[DBG] iotx_mqtt_client.c:1853 [2021-04-27 15:40:01.074] network is disconnected!
[DBG] Iot.c:24 [2021-04-27 15:40:01.075] MQTT disconnect.
[DBG] HAL_TLS_mbedtls.c:713 [2021-04-27 15:40:01.091] ssl_disconnect
[DBG] iotx_mqtt_client.c:1492 [2021-04-27 15:40:01.092] state = 4
[DBG] iotx_mqtt_client.c:1599 [2021-04-27 15:40:01.103] error occur rc=-27
[DBG] iotx_mqtt_client.c:1492 [2021-04-27 15:40:01.103] state = 4
[DBG] iotx_mqtt_client.c:3099 [2021-04-27 15:40:01.116] mqtt client state is error,state = 4
[DBG] iotx_mqtt_client.c:1743 [2021-04-27 15:40:01.128] Waiting to reconnect...
[DBG] iotx_mqtt_client.c:1839 [2021-04-27 15:40:01.229] reconnect network fail, rc = -1
[DBG] iotx_mqtt_client.c:1492 [2021-04-27 15:40:01.230] state = 4
[DBG] iotx_mqtt_client.c:1599 [2021-04-27 15:40:01.240] error occur rc=-27
[DBG] iotx_mqtt_client.c:1492 [2021-04-27 15:40:01.241] state = 4
[DBG] CIot.cpp:448 [2021-04-27 15:40:01.250] =============================================Yield time=122ms
[DBG] iotx_mqtt_client.c:1743 [2021-04-27 15:40:01.253] Waiting to reconnect...
[DBG] iotx_mqtt_client.c:1839 [2021-04-27 15:40:01.354] reconnect network fail, rc = -1
[DBG] iotx_mqtt_client.c:1492 [2021-04-27 15:40:01.355] state = 4
[DBG] iotx_mqtt_client.c:1599 [2021-04-27 15:40:01.365] error occur rc=-27
[DBG] iotx_mqtt_client.c:1492 [2021-04-27 15:40:01.366] state = 4
[DBG] CIot.cpp:448 [2021-04-27 15:40:01.377] =============================================Yield time=124ms
[DBG] iotx_mqtt_client.c:1743 [2021-04-27 15:40:01.381] Waiting to reconnect...
[DBG] iotx_mqtt_client.c:1839 [2021-04-27 15:40:01.481] reconnect network fail, rc = -1
[DBG] iotx_mqtt_client.c:1492 [2021-04-27 15:40:01.482] state = 4
[DBG] iotx_mqtt_client.c:1599 [2021-04-27 15:40:01.493] error occur rc=-27
[DBG] iotx_mqtt_client.c:1492 [2021-04-27 15:40:01.493] state = 4
[DBG] CIot.cpp:448 [2021-04-27 15:40:01.502] =============================================Yield time=121ms
[DBG] iotx_mqtt_client.c:1743 [2021-04-27 15:40:01.506] Waiting to reconnect...
[DBG] iotx_mqtt_client.c:1839 [2021-04-27 15:40:01.606] reconnect network fail, rc = -1
[DBG] iotx_mqtt_client.c:1492 [2021-04-27 15:40:01.617] state = 4
[DBG] iotx_mqtt_client.c:1599 [2021-04-27 15:40:01.628] error occur rc=-27
[DBG] iotx_mqtt_client.c:1492 [2021-04-27 15:40:01.628] state = 4
[DBG] CIot.cpp:448 [2021-04-27 15:40:01.637] =============================================Yield time=132ms
[DBG] iotx_mqtt_client.c:1743 [2021-04-27 15:40:01.641] Waiting to reconnect...
[DBG] iotx_mqtt_client.c:1839 [2021-04-27 15:40:01.741] reconnect network fail, rc = -1
[DBG] iotx_mqtt_client.c:1492 [2021-04-27 15:40:01.742] state = 4
[DBG] iotx_mqtt_client.c:1599 [2021-04-27 15:40:01.753] error occur rc=-27
[DBG] iotx_mqtt_client.c:1492 [2021-04-27 15:40:01.753] state = 4
[DBG] CIot.cpp:448 [2021-04-27 15:40:01.762] =============================================Yield time=121ms
[DBG] iotx_mqtt_client.c:1743 [2021-04-27 15:40:01.766] Waiting to reconnect...
[DBG] iotx_mqtt_client.c:1839 [2021-04-27 15:40:01.866] reconnect network fail, rc = -1
[DBG] iotx_mqtt_client.c:1492 [2021-04-27 15:40:01.877] state = 4
[DBG] iotx_mqtt_client.c:1599 [2021-04-27 15:40:01.888] error occur rc=-27
[DBG] iotx_mqtt_client.c:1492 [2021-04-27 15:40:01.889] state = 4
[DBG] CIot.cpp:448 [2021-04-27 15:40:01.897] =============================================Yield time=131ms
[DBG] iotx_mqtt_client.c:1743 [2021-04-27 15:40:01.901] Waiting to reconnect...
[DBG] iotx_mqtt_client.c:1839 [2021-04-27 15:40:02.002] reconnect network fail, rc = -1
[DBG] iotx_mqtt_client.c:1492 [2021-04-27 15:40:02.003] state = 4
[DBG] iotx_mqtt_client.c:1599 [2021-04-27 15:40:02.013] error occur rc=-27
[DBG] iotx_mqtt_client.c:1492 [2021-04-27 15:40:02.014] state = 4
[DBG] CIot.cpp:448 [2021-04-27 15:40:02.023] =============================================Yield time=122ms
[DBG] iotx_mqtt_client.c:1743 [2021-04-27 15:40:02.027] Waiting to reconnect...
[DBG] iotx_mqtt_client.c:1839 [2021-04-27 15:40:02.128] reconnect network fail, rc = -1
[DBG] iotx_mqtt_client.c:1492 [2021-04-27 15:40:02.128] state = 4
[DBG] iotx_mqtt_client.c:1599 [2021-04-27 15:40:02.139] error occur rc=-27
[DBG] iotx_mqtt_client.c:1492 [2021-04-27 15:40:02.140] state = 4
[DBG] CIot.cpp:448 [2021-04-27 15:40:02.148] =============================================Yield time=121ms
[DBG] iotx_mqtt_client.c:1743 [2021-04-27 15:40:02.152] Waiting to reconnect...
[DBG] iotx_mqtt_client.c:1750 [2021-04-27 15:40:02.153] start to reconnect
[DBG] HAL_TLS_mbedtls.c:729 [2021-04-27 15:40:02.153] handle is NULL
[... cut ...]
[DBG] iotx_mqtt_client.c:778 [2021-04-27 15:40:02.915] connect params: MQTTVersion=4, clientID=(此处隐藏)|timestamp=2524608000000,_v=sdk-c-3.0.1,securemode=2,signmethod=hmacsha256,lan=C,gw=0,ext=0|, keepAliveInterval=60, username=(此处隐藏)
[DBG] iotx_mqtt_client.c:821 [2021-04-27 15:40:02.980] mqtt connect success!
[DBG] iotx_mqtt_client.c:1843 [2021-04-27 15:40:02.981] network is reconnected!
[DBG] Iot.c:28 [2021-04-27 15:40:02.981] MQTT reconnect.
[DBG] CIot.cpp:448 [2021-04-27 15:40:03.002] =============================================Yield time=850ms