aliyun/iotkit-embedded #101: ARMCC 编译SDK

lingyun0129, 2018-09-29:
因GNU编译工具编译的SDK，在mtk平台出现链接错误。故想问能否用armcc来编译SDK？该如何操作，能否给予指导？

edwardyangx, 2018-11-23:
可以尝试关闭编译选项 -Os 之后再重新链接

pansila, 2018-11-26:
什么链接错误，我可以用MDK链接GCC编出来的库，注意选择正确的CFLAGS，比如-fshort-wchar，-mthumb，-mfloat-abi，另外一些newlib的libc库函数底层调用需要自己封装，改用mircolib的或者自己实现。

zyh2591, 2019-01-08:
请问楼上采用的是arm-none-eabi-gcc还是GNU Make + gcc 编译链接到MDK。有没有什么指导的网站或者文档分享一下，很困惑

superdsp, 2019-01-23:
请问用MDK5 ARM Complier来链接 SDK， 使用arm-none-eabi-gcc 编译出的库文件
[image]，
[image]
链接后出问题，
T407_SOM\T407_SOM.axf: Error: L6242E: Cannot link object sdk-impl.o as its attributes are incompatible with the image attributes.
   ... wchart-16 clashes with wchart-32.
T407_SOM\T407_SOM.axf: Error: L6242E: Cannot link object mqtt_client.o as its attributes are incompatible with the image attributes.
   ... wchart-16 clashes with wchart-32.
T407_SOM\T407_SOM.axf: Error: L6242E: Cannot link object mem_stats.o as its attributes are incompatible with the image attributes.
   ... wchart-16 clashes with wchart-32.
请问大家有办法解决么？

pansila, 2019-02-11:
@zyh2591 arm-none-eabi-gcc参照sdk里面提高的几个config改下就行了

pansila, 2019-02-11:
@superdsp 这是宽字符问题，要给gcc指定宽字符的大小，-fshort-wchar

另附上我遇到的问题做的patch。
[board.zip](https://github.com/aliyun/iotkit-embedded/files/2850333/board.zip)