aliyun/iotkit-embedded #103: OTA升级提示fail to download，如何解决

xwdreamer, 2018-10-25:
在控制台操作升级固件，在设备端提示如下日志：
——————————————————————————
[dbg] httpclient_common(785): host: 'iotx-ota.oss-cn-shanghai.aliyuncs.com', port: 443
LINUXSOCK 77 HAL_TCP_Establish() | establish tcp connection with server(host=iotx-ota.oss-cn-shanghai.aliyuncs.com port=443)
LINUXSOCK 116 HAL_TCP_Establish() | success to establish tcp, fd=5
[dbg] httpclient_send_header(317): REQUEST (Length: 328 Bytes)
> GET /ota/4ab1305d8620198918b921eb5b330bdb/cjnng5uee0001396wk1er8vjm.zip?Expires=1540440613&OSSAccessKeyId=cS8uRRy54RszYWna&Signature=0OE%2B2JqLwBdzyWn6dNVBoLbjkA0%3D HTTP/1.1
> Host: iotx-ota.oss-cn-shanghai.aliyuncs.com
> Accept: text/html,application/xhtml+xml,application/xml;q=0.9,*/*;q=0.8
> Accept-Encoding: gzip, deflate
> 
[dbg] httpclient_send_header(322): Written 328 bytes
[inf] httpclient_recv(384): 32 bytes has been read
[dbg] httpclient_recv_response(760): RESPONSE (Length: 32 Bytes)
< HTTP/1.1 400 Bad Request
< Server
[wrn] httpclient_response_parse(643): Response code 400
[dbg] httpclient_response_parse(646): Reading headers: HTTP/1.1 400 Bad Request
[inf] httpclient_recv(384): 32 bytes has been read
[inf] httpclient_recv(384): 32 bytes has been read
[inf] httpclient_recv(384): 32 bytes has been read
[inf] httpclient_recv(384): 32 bytes has been read
[dbg] httpclient_retrieve_content(428): Current data: <!DOCTYPE H
[dbg] httpclient_retrieve_content(534): Total-Payload: 281 Bytes; Read: 11 Bytes
[inf] httpclient_recv(384): 270 bytes has been read
[dbg] httpclient_retrieve_content(589): no more (content-length)
[inf] httpclient_common(828): close http channel
[dbg] httpclient_close(774): client disconnected
mqtt_client|247 :: ota fetch fail
——————————————————————————
我自己拼装上面的url，又是可以下载到固件的
https://iotx-ota.oss-cn-shanghai.aliyuncs.com/ota/4ab1305d8620198918b921eb5b330bdb/cjnng5uee0001396wk1er8vjm.zip?Expires=1540440613&OSSAccessKeyId=cS8uRRy54RszYWna&Signature=0OE%2B2JqLwBdzyWn6dNVBoLbjkA0%3D

问题：
为何会提示fail to down

[image]

[image]

xwdreamer, 2018-10-25:
sdk版本：RELEASED_V2_10_20180331.zip

xwdreamer, 2018-10-25:
通过wireshark抓包，完整的返回信息如下
[image]

LeeSSXX, 2019-04-16:
用2.3.0吧