aliyun/iotkit-embedded #125: 求助：xtensa-lx106-elf-nm libiot_sdk.a: no symbols

deemstone, 2019-02-08:
make reconfig  #  select 1 (config.esp8266.aos)
make

xtensa-lx106-elf-nm output/release/lib/libiot_sdk.a

Output: 
> utils_base64.o:
> xtensa-lx106-elf-nm: utils_base64.o: no symbols
> 
> utils_hmac.o:
> xtensa-lx106-elf-nm: utils_hmac.o: no symbols
> 
> utils_md5.o:
> xtensa-lx106-elf-nm: utils_md5.o: no symbols
> 
> utils_sha1.o:
> xtensa-lx106-elf-nm: utils_sha1.o: no symbols
> 
> utils_sha256.o:
> xtensa-lx106-elf-nm: utils_sha256.o: no symbols
> ......

所有的 *.o 文件都是 `no symbols`， 导致 [esp-aliyun](https://github.com/espressif/esp-aliyun)项目编译失败：

> iotkit-embedded/output/release/lib/libiot_sdk.a: error adding symbols: Archive has no index; run ranlib to add one

.config :

```makefile
CONFIG_src/tools/linkkit_tsl_convert = y
CONFIG_src/services/subdev = y
CONFIG_src/services/shadow = y
CONFIG_src/services/ota = y
CONFIG_src/services/mdal/sal = y
CONFIG_src/services/mdal/mal = y
CONFIG_src/services/linkkit/dm = y
CONFIG_src/services/linkkit/dev_reset = y
CONFIG_src/services/linkkit/cm = y
CONFIG_src/services/http2_stream = y
CONFIG_src/services/dev_bind = y
CONFIG_src/services/awss = y
CONFIG_src/sdk-impl = y
CONFIG_src/ref-impl/tls = y
CONFIG_src/ref-impl/hal = y
CONFIG_src/protocol/mqtt = y
CONFIG_src/protocol/http2 = y
CONFIG_src/protocol/http = y
CONFIG_src/protocol/coap/local = y
CONFIG_src/protocol/coap/cloud = y
CONFIG_src/protocol/alcs = y
CONFIG_src/infra/utils = y
CONFIG_src/infra/system = y
CONFIG_src/infra/log = y
CONFIG_examples = y
# Automatically Generated Section End

# VENDOR :   esp8266
# MODEL  :   aos
CONFIG_ENV_CFLAGS   += \
    -DBOARD_ESP8266 -u call_user_start \
    -fno-inline-functions \
    -ffunction-sections \
    -fdata-sections \
    -mlongcalls \
    -DESPOS_FOR_ESP8266 -Wl,-static \
    -DXT_USE_THREAD_SAFE_CLIB=0 \

CONFIG_ENV_CFLAGS   += \
    -Os \
    -DCONFIG_HTTP_AUTH_TIMEOUT=500 \
    -DCONFIG_MID_HTTP_TIMEOUT=500 \
    -DCONFIG_GUIDER_AUTH_TIMEOUT=500 \
    -DCONFIG_MQTT_TX_MAXLEN=640 \
    -DCONFIG_MQTT_RX_MAXLEN=1200 \

CONFIG_src/ref-impl/tls         :=
CONFIG_src/ref-impl/hal         :=
CONFIG_examples                 :=
CONFIG_tests                    :=
CONFIG_src/tools/linkkit_tsl_convert :=

CROSS_PREFIX        := xtensa-lx106-elf-
```

27-301, 2019-02-28:
哪个平台下编译？mac 吗？

deemstone, 2019-02-28:
> 哪个平台下编译？mac 吗？

是的 Mac 系统，是什么原因？

27-301, 2019-03-01:
查了下，发现原因可能是 mac 系统下降 libiot_sdk.a strip 后产生的问题。

文件：https://github.com/aliyun/iotkit-embedded/blob/master/build-rules/settings.mk#L6-L12

mac 下 strip 默认没有参数：
```shell
ifeq (Darwin,$(strip $(shell uname)))
SED             := gsed
STRIP_DBGOPT    :=
else
SED             := sed
STRIP_DBGOPT    := --strip-debug
endif
```
使用和非 mac 系统一致的参数可以解决该问题。

阿里的同学能否确认一下，能否将 mac 下也加入该参数。

27-301, 2019-03-01:
@deemstone https://github.com/espressif/esp-aliyun 已更新，可以支持 MacOS 下的编译。

请按以下 steps：
1. 8266 SDK 请 checkout release/v3.1 分支
2. esp-aliyun 目录下 `cp -r patch/* iotkit-embedded`
3. 编译 libiot_sdk.a
4. 编译应用

当前已支持 awss，也可以直接打通 飞燕 平台，可以参看 examples/solutions/smart_light，esp32 & esp8266 均已支持。但可能还会有一些小 bug，欢迎去提 issue。

deemstone, 2019-03-02:
@27-301 可以正常编译了，棒！