aliyun/iotkit-embedded #128: tag v2.3.0 2个月之前还能编译通过，这两个月发现无法编译通过了

LeeSSXX, 2019-03-06:
我的系统Ubuntu 16.04 64位
之前我检出来的tag 2.3.0的hash是 ad4d60be1dd44b0292d55938dcdc27a1c7d08068
现在检出来的tag 2.3.0的hash是 e5cd733dfb0c9b802bce39c54bcbc98bdf4b1bf3

但现在无法编译通过，出现如下错误
```
/opt/develCode/agent-proxy/third_party/iotkit-embedded/.O/usr/lib/libiot_sdk.a(iotx_cm_mqtt.o): In function `iotx_cloud_conn_mqtt_event_handle':
/opt/develCode/agent-proxy/third_party/iotkit-embedded/src/services/linkkit/cm/iotx_cm_mqtt.c:210: undefined reference to `awss_enrollee_checkin'
/opt/develCode/agent-proxy/third_party/iotkit-embedded/src/services/linkkit/cm/iotx_cm_mqtt.c:212: undefined reference to `awss_report_enrollee_reply'
/opt/develCode/agent-proxy/third_party/iotkit-embedded/src/services/linkkit/cm/iotx_cm_mqtt.c:214: undefined reference to `awss_get_cipher_reply'
/opt/develCode/agent-proxy/third_party/iotkit-embedded/src/services/linkkit/cm/iotx_cm_mqtt.c:216: undefined reference to `awss_online_switchap'
collect2: error: ld returned 1 exit status
/opt/develCode/agent-proxy/third_party/iotkit-embedded/.O/.one_makefile:475: recipe for target '/opt/develCode/agent-proxy/third_party/iotkit-embedded/.O/usr/bin/ota-example-mqtt' failed
make[1]: *** [/opt/develCode/agent-proxy/third_party/iotkit-embedded/.O/usr/bin/ota-example-mqtt] Error 1
make[1]: *** Waiting for unfinished jobs....
/opt/develCode/agent-proxy/third_party/iotkit-embedded/.O/usr/lib/libiot_sdk.a(iotx_cm_mqtt.o): In function `iotx_cloud_conn_mqtt_event_handle':
/opt/develCode/agent-proxy/third_party/iotkit-embedded/src/services/linkkit/cm/iotx_cm_mqtt.c:210: undefined reference to `awss_enrollee_checkin'
/opt/develCode/agent-proxy/third_party/iotkit-embedded/src/services/linkkit/cm/iotx_cm_mqtt.c:212: undefined reference to `awss_report_enrollee_reply'
/opt/develCode/agent-proxy/third_party/iotkit-embedded/src/services/linkkit/cm/iotx_cm_mqtt.c:214: undefined reference to `awss_get_cipher_reply'
/opt/develCode/agent-proxy/third_party/iotkit-embedded/src/services/linkkit/cm/iotx_cm_mqtt.c:216: undefined reference to `awss_online_switchap'
collect2: error: ld returned 1 exit status
/opt/develCode/agent-proxy/third_party/iotkit-embedded/.O/.one_makefile:442: recipe for target '/opt/develCode/agent-proxy/third_party/iotkit-embedded/.O/usr/bin/mqtt-example' failed
make[1]: *** [/opt/develCode/agent-proxy/third_party/iotkit-embedded/.O/usr/bin/mqtt-example] Error 1
/opt/develCode/agent-proxy/third_party/iotkit-embedded/.O/usr/lib/libiot_sdk.a(iotx_cm_mqtt.o): In function `iotx_cloud_conn_mqtt_event_handle':
/opt/develCode/agent-proxy/third_party/iotkit-embedded/src/services/linkkit/cm/iotx_cm_mqtt.c:210: undefined reference to `awss_enrollee_checkin'
/opt/develCode/agent-proxy/third_party/iotkit-embedded/src/services/linkkit/cm/iotx_cm_mqtt.c:212: undefined reference to `awss_report_enrollee_reply'
/opt/develCode/agent-proxy/third_party/iotkit-embedded/src/services/linkkit/cm/iotx_cm_mqtt.c:214: undefined reference to `awss_get_cipher_reply'
/opt/develCode/agent-proxy/third_party/iotkit-embedded/src/services/linkkit/cm/iotx_cm_mqtt.c:216: undefined reference to `awss_online_switchap'
collect2: error: ld returned 1 exit status
/opt/develCode/agent-proxy/third_party/iotkit-embedded/.O/.one_makefile:425: recipe for target '/opt/develCode/agent-proxy/third_party/iotkit-embedded/.O/usr/bin/mqtt-example-rrpc' failed
make[1]: *** [/opt/develCode/agent-proxy/third_party/iotkit-embedded/.O/usr/bin/mqtt-example-rrpc] Error 1
/opt/develCode/agent-proxy/third_party/iotkit-embedded/.O/usr/lib/libiot_sdk.a(iotx_cm_mqtt.o): In function `iotx_cloud_conn_mqtt_event_handle':
/opt/develCode/agent-proxy/third_party/iotkit-embedded/src/services/linkkit/cm/iotx_cm_mqtt.c:210: undefined reference to `awss_enrollee_checkin'
/opt/develCode/agent-proxy/third_party/iotkit-embedded/src/services/linkkit/cm/iotx_cm_mqtt.c:212: undefined reference to `awss_report_enrollee_reply'
/opt/develCode/agent-proxy/third_party/iotkit-embedded/src/services/linkkit/cm/iotx_cm_mqtt.c:214: undefined reference to `awss_get_cipher_reply'
/opt/develCode/agent-proxy/third_party/iotkit-embedded/src/services/linkkit/cm/iotx_cm_mqtt.c:216: undefined reference to `awss_online_switchap'
collect2: error: ld returned 1 exit status
/opt/develCode/agent-proxy/third_party/iotkit-embedded/.O/.one_makefile:459: recipe for target '/opt/develCode/agent-proxy/third_party/iotkit-embedded/.O/usr/bin/mqtt-example-multithread' failed
make[1]: *** [/opt/develCode/agent-proxy/third_party/iotkit-embedded/.O/usr/bin/mqtt-example-multithread] Error 1
/opt/develCode/agent-proxy/third_party/iotkit-embedded/build-rules/_rules-submods.mk:5: recipe for target 'sub-mods' failed
make: *** [sub-mods] Error 2
root@Develop:/opt/develCode/agent-proxy/third_party/iotkit-embedded#
```

你们tag 打过之后也要变吗？？

shellohunter, 2019-03-08:
刚下了一版，也编译失败了，错误还不一样。
代码管理有点混乱。从branch上看版本是2.3.0, 但tag打的是2.30，hash跟你的还不一样。我晕

LeeSSXX, 2019-03-08:
> 刚下了一版，也编译失败了，错误还不一样。
> 代码管理有点混乱。从branch上看版本是2.3.0, 但tag打的是2.30，hash跟你的还不一样。我晕

make menuconfig 关闭FEATURE_DEV_BIND_ENABLED ,可以编译通过

shellohunter, 2019-03-08:
话说，你的tag为啥跟我的不一样。。。
```
[xxxx@localhost iotkit-embedded]$ git tag --list
RELEASED_V2.00
RELEASED_V2.01
RELEASED_V2.02
RELEASED_V2.03
RELEASED_V2.20
RELEASED_V2.21
RELEASED_V2.30
[xxxx@localhost iotkit-embedded]$ git show RELEASED_V2.30
commit e941e190cc709654b711827e4192a97028deac59
Author: Yang, Xiao <yusheng.yx@alibaba-inc.com>
Date:   Wed Nov 21 11:37:31 2018 +0800

    [mingw32] fix build error when generate Windows version SDK

    Signed-off-by: Yang, Xiao <yusheng.yx@alibaba-inc.com>
```

closed 2019-03-08 by LeeSSXX

edwardyangx, 2019-03-15:
v2.3.0不是tag, 是branch的名字