aliyun/iotkit-embedded #131: DM_MSG_PROPERTY_SET无法获取object数据

Torchwoods, 2019-03-12:
版本：V2.3.0
make.config配置:
`
#
# Automatically generated file; DO NOT EDIT.
# Main Menu
#

#
# Configure Link Kit SDK for IoT Embedded Devices
#
FEATURE_SRCPATH="."
FEATURE_MQTT_COMM_ENABLED=y

#
# MQTT Configurations
#
FEATURE_MQTT_DIRECT=y
# FEATURE_MQTT_SHADOW is not set
# FEATURE_MQTT_LOGPOST is not set
FEATURE_DEVICE_MODEL_ENABLED=y

#
# Device Model Configurations
#
FEATURE_DEVICE_MODEL_GATEWAY=y
FEATURE_ALCS_ENABLED=y

#
# ALCS Configurations
#
FEATURE_ALCS_CLIENT_ENABLED=y
FEATURE_ALCS_SERVER_ENABLED=y
FEATURE_DEPRECATED_LINKKIT=y
FEATURE_DEV_BIND_ENABLED=y
# FEATURE_COAP_COMM_ENABLED is not set
FEATURE_OTA_ENABLED=y
# FEATURE_HTTP2_COMM_ENABLED is not set
# FEATURE_HTTP_COMM_ENABLED is not set
FEATURE_SUPPORT_TLS=y
FEATURE_SAL_ENABLED=y

#
# SAL Configurations
#
FEATURE_SAL_PACKET_SEND_MODE_ASYNC=y
FEATURE_SAL_UDP_CLIENT_ENABLED=y
FEATURE_WIFI_PROVISION_ENABLED=y

#
# AWSS Configurations
#
FEATURE_AWSS_SUPPORT_SMARTCONFIG=y
FEATURE_AWSS_SUPPORT_PHONEASAP=y
FEATURE_AWSS_SUPPORT_ROUTER=y
FEATURE_AWSS_SUPPORT_ZEROCONFIG=y
`
问题：
为了兼容V2.2.1，在V2.3.0版本上使用老版本接口实现网关功能，在网关里面注册了一个Zigbee开关设备，通过后端开关设备的调试界面给网关发送set命令控制开关状态，数据发送到网关后，并没有调用到子设备注册的linkkit_cbs里面的set_property回调函数
`
[dbg] CoAPRequestMessage_handle(730): Request path is /dev/a1j22B9pjJ0/sIfgUD2omYCseZGD4ilp/core/service/auth
[dbg] iotx_mc_cycle(1821): PUBLISH
[inf] iotx_mc_handle_recv_PUBLISH(1574): Downstream Topic: '/sys/a1j22B9pjJ0/sIfgUD2omYCseZGD4ilp/thing/service/property/set'
[inf] iotx_mc_handle_recv_PUBLISH(1575): Downstream Payload:

< {
<     "method": "thing.service.property.set",
<     "id": "280608376",
<     "params": {
<         "PowerSwitch_1": 0
<     },
<     "version": "1.0.0"
< }

[dbg] iotx_mc_handle_recv_PUBLISH(1580):         Packet Ident : 00000000
[dbg] iotx_mc_handle_recv_PUBLISH(1581):         Topic Length : 64
[dbg] iotx_mc_handle_recv_PUBLISH(1585):           Topic Name : /sys/a1j22B9pjJ0/sIfgUD2omYCseZGD4ilp/thing/service/property/set
[dbg] iotx_mc_handle_recv_PUBLISH(1588):     Payload Len/Room : 103 / 111
[dbg] iotx_mc_handle_recv_PUBLISH(1589):       Receive Buflen : 180
[dbg] iotx_mc_handle_recv_PUBLISH(1600): delivering msg ...
[dbg] iotx_mc_deliver_message(1307): topic be matched
[inf] dm_msg_proc_thing_service_property_set(134): thing/service/property/set
[dbg] dm_msg_request_parse(130): Current Request Message ID: 280608376
[dbg] dm_msg_request_parse(131): Current Request Message Version: 1.0.0
[dbg] dm_msg_request_parse(132): Current Request Message Method: thing.service.property.set
[dbg] dm_msg_request_parse(133): Current Request Message Params: {"PowerSwitch_1":0}
[inf] dm_msg_property_set(2194): Property Set, Size: 1
[dbg] _dm_msg_set_object(2074): Current Key: PowerSwitch_1, Value: 0
[dbg] _dm_msg_set_object(2077): new_key_len: 14
[dbg] _dm_msg_set_object(2088): New Key: PowerSwitch_1
[inf] dm_msg_property_set(2201): _dm_msg_set_object failed
[inf] dm_msg_response(262): Send URI: /sys/a1j22B9pjJ0/sIfgUD2omYCseZGD4ilp/thing/service/property/set_reply, Payload: {"id":"280608376","code":400,"data":{}}
[inf] MQTTPublish(514): Upstream Topic: '/sys/a1j22B9pjJ0/sIfgUD2omYCseZGD4ilp/thing/service/property/set_reply'
[inf] MQTTPublish(515): Upstream Payload:

> {
>     "id": "280608376",
>     "code": 400,
>     "data": {
>     }
> }

[inf] dm_client_publish(121): Publish Result: 0
[zbSocUartProcessHandler:267]:ByteToRead = 16
`
调试后发现是dm_manager.c中的dm_shw_get_property_data函数里面shadow参数为NULL，导致参数验证错误，在代码中也没有发现这个参数在哪里初始化，分配的空间。

Bloodance, 2019-03-13:
tsl是本地还是云端获取的？

Torchwoods, 2019-03-17:
云端获取

Bloodance, 2019-03-22:
@Torchwoods 建议改为本地设置

edwardyangx, 2019-03-24:
你好以上问题已在

https://github.com/aliyun/iotkit-embedded/commit/0a018337f1d65e988015103adfda74c266df1e56

修复, 请更新代码, 感谢反馈!